UMSAE-Formula-Electric/electrical #30: Sam Sponsorship Initial Contacts

MitchHewitt, 2025-09-14:
## Task Description
Within this issue please leave comments for each company contacted

## Expected Outcome
Please leave the company name with a copy of the email you sent to the company in an individual comment below this issue.

## Additional Context
CHATGPT PROMPT FOR EMAILS

You are an assistant helping me draft personalized sponsorship outreach emails. I will provide a company website link. Your task is to:

Review the company’s mission, values, and recent projects.

Write a professional outreach email that highlights how their values align with ours.

Keep the email specific and approachable, not generic.

Avoid filler phrases like “I hope this message finds you well” or “best regards.”

End with a clear call to action (e.g., scheduling a call or discussing sponsorship opportunities).

About Us – UMSAE Formula Electric (Polar Bear Racing):
We are the University of Manitoba’s Formula SAE Electric team, a student-run group that designs and builds a fully electric, formula-style racecar each year for international competition. Based in Winnipeg, we focus on sustainability, innovation, and hands-on engineering. Sponsorship supports design, manufacturing, and testing, helping us push EV technology forward while preparing the next generation of engineers.

Your task:
Once I provide a company link, draft a concise, tailored outreach email that:

Opens by recognizing the company’s achievements or values.

Introduces our team and mission.

Shows the alignment between us and the company.

Explains the benefit of partnering.

Closes with a clear next step.

I will paste the company link next.

samosh3, 2025-10-26:
Company: Antec Controls
Contact: Devon Sowiak - Territory Manager devon.sowiak@anteccontrols.com linkdin: https://www.linkedin.com/in/devon-sowiak-978160223/

Hi Devon, 

My name is Samantha O'Shea and I am reaching out on behalf of UMSAE Formula Electric (Polar Bear Racing) at the University of Manitoba. Our student-run team designs and builds a fully electric, formula-style race car each year for international competition. Based in Winnipeg, we prioritize sustainability, hands-on engineering and pushing cutting-edge EV technology.

We are interested in a partnership with Antec Controls to support the continued development of our local team. As a Winnipeg-based organization developing global impact in engineering, you and our team share a foundation in innovation, local engineering talent and real-world performance.
Partnering with UMSAE Formula Electric would offer Antec Controls several benefits:
Your brand will gain visibility at national and international Formula SAE Electric events as well as in university and media outreach.
Access to a pool of skilled student engineers proficient in electric-vehicle systems, sustainable design and hands-on build experience.
If this aligns with Antec Controls’ interests, I’d be glad to continue the discussion and share our partnership package along with our Dean’s letter of approval.
Looking forward to hearing your thoughts,

samosh3, 2025-10-26:
Company: Smith&Andersen
Contact: Nick Chu - Principal Engineer "nick.chu@smithandandersen.com" 

Hi Nick, 

I hope this email finds you well! My name is Samantha O'Shea and I was a Junior Electrical Designer at the Winnipeg office over the summer of 2025. How have things been at Smith+Andersen?

I’m currently a member of the University of Manitoba Society of Automotive Engineers (UMSAE), an engineering student group at the University of Manitoba. UMSAE is made up of four teams: Aero, Baja, Formula, and Formula Electric. I am the Electrical Manufacturing lead on the Formula Electric team, where we focus on Designing and Manufacturing a fully electric formula-style racecar. 

All of our work is funded entirely through sponsorships and donations, and we are reaching out to S+A to ask if you’d be interested in supporting us. A sponsorship from S+A would not only help us achieve our goals but also provide a unique opportunity for the company. Sponsorship benefits include having your logo prominently displayed on our car as well as included in sponsorship posts on our social media.

I think this would be a great chance for S+A to build recognition in Winnipeg. UMSAE has over 250 engineering students and with a monetary sponsorship S+A would be able to get their name in the minds of those applying for Co-op positions or for those graduating looking for full time work.

If you're interested, let me know and we can discuss further. We also have a sponsorship package that I can send over to you that goes into more detail of all the sponsorship tiers.

closed 2025-11-23 by MitchHewitt