UMSAE-Formula-Electric/electrical #31: Taylor Initial Sponsorship requests

MitchHewitt, 2025-09-14:
## Task Description
Within this issue please leave comments for each company contacted

## Expected Outcome
Please leave the company name with a copy of the email you sent to the company in an individual comment below this issue.

## Additional Context
CHATGPT PROMPT FOR EMAILS

You are an assistant helping me draft personalized sponsorship outreach emails. I will provide a company website link. Your task is to:

Review the company’s mission, values, and recent projects.

Write a professional outreach email that highlights how their values align with ours.

Keep the email specific and approachable, not generic.

Avoid filler phrases like “I hope this message finds you well” or “best regards.”

End with a clear call to action (e.g., scheduling a call or discussing sponsorship opportunities).

About Us – UMSAE Formula Electric (Polar Bear Racing):
We are the University of Manitoba’s Formula SAE Electric team, a student-run group that designs and builds a fully electric, formula-style racecar each year for international competition. Based in Winnipeg, we focus on sustainability, innovation, and hands-on engineering. Sponsorship supports design, manufacturing, and testing, helping us push EV technology forward while preparing the next generation of engineers.

Your task:
Once I provide a company link, draft a concise, tailored outreach email that:

Opens by recognizing the company’s achievements or values.

Introduces our team and mission.

Shows the alignment between us and the company.

Explains the benefit of partnering.

Closes with a clear next step.

I will paste the company link next.

TaylorNelson123, 2025-09-15:
Organization: Bill Nye the Science Guy
Contact: Christine Sposari - Assistant to Bill Nye (bnsg@billnye.com)

Email:

Hello Bill Nye Team,

I’m writing on behalf of UMSAE Formula Electric at the University of Manitoba. We are a student-run engineering team designing, building, and testing a fully electric formula-style racecar each year for international competition. Our work combines hands-on engineering, sustainability, and cutting-edge EV technology, all to prepare the next generation of engineers.

There is strong alignment between what you promote and what we do. Just as you engage audiences by explaining and applying scientific processes, our team brings engineering principles to life through real-world challenges such as energy efficiency, battery systems, and aerodynamics. Our focus on sustainability reflects your advocacy for climate action, as we build electric vehicles that demonstrate practical solutions for reducing emissions. Equally important, we share a commitment to education, providing students with opportunities to turn theory into practice, embrace experimentation, and learn through both success and failure.

More information about our organization can be found here www.umsae.com/formula-e
If this opportunity interests you, I’d be glad to continue the conversation over email and share more details about our team, sponsorship options, and ways we could collaborate.

TaylorNelson123, 2025-10-05:
Organization: Hoonigan
Contact: The website
Email:

Hello,

My name is Taylor, and I’m an electrical engineering student at the University of Manitoba. I’m part of UMSAE, which operates under SAE International, and I currently serve as a lead on the Formula Electric team. Our team designs, builds, and races a fully electric formula-style car each year.

I’m reaching out to see if your organization would be interested in a sponsorship opportunity with our team.

TaylorNelson123, 2025-10-19:
Organization: Winnipeg Supply 
Contact: The Website
Email:
I’m part of UMSAE Formula Electric at the University of Manitoba, where we design and build an all-electric race car for international competition. Your focus on efficient heating and cooling aligns with our commitment to sustainability and innovation. We’d love to discuss a sponsorship or partnership opportunity with Winnipeg Supply.

TaylorNelson123, 2025-10-26:
Organization: Sysco
Contact: info@corp.sysco.ca
Email
Hello,

I’m writing on behalf of the student team at the UMSAE Formula Electric (Polar Bear Racing) based at the University of Manitoba. We design and build a fully electric, formula-style racecar each year for international competition. Situated in Winnipeg, our work centers on sustainability, hands-on engineering and innovation, preparing the next generation of engineers.
Here’s why I believe our paths naturally align:
Your emphasis on caring for community mirrors our commitment to engaging students and the broader Winnipeg region in sustainable tech.
Your pursuit of responsible growth and high standards reflects our insistence on engineering excellence and sustainable innovation.
Your local operations mean our collaboration would deliver tangible impact in the same neighborhood where we both operate.
We would love to explore a collaboration that highlights what both of us are working toward in sustainability and innovation. I’d be happy to share more details about our team, our upcoming competition season or a few ideas for how we could work together

Thanks,
Taylor Nelson
Electrical Engineering Student
University  of Manitoba