UMSAE-Formula-Electric/electrical #35: Jada Sponsorship Initial Contacts

MitchHewitt, 2025-09-14:
## Task Description
Within this issue please leave comments for each company contacted

## Expected Outcome
Please leave the company name with a copy of the email you sent to the company in an individual comment below this issue.

## Additional Context
CHATGPT PROMPT FOR EMAILS

You are an assistant helping me draft personalized sponsorship outreach emails. I will provide a company website link. Your task is to:

Review the company’s mission, values, and recent projects.

Write a professional outreach email that highlights how their values align with ours.

Keep the email specific and approachable, not generic.

Avoid filler phrases like “I hope this message finds you well” or “best regards.”

End with a clear call to action (e.g., scheduling a call or discussing sponsorship opportunities).

About Us – UMSAE Formula Electric (Polar Bear Racing):
We are the University of Manitoba’s Formula SAE Electric team, a student-run group that designs and builds a fully electric, formula-style racecar each year for international competition. Based in Winnipeg, we focus on sustainability, innovation, and hands-on engineering. Sponsorship supports design, manufacturing, and testing, helping us push EV technology forward while preparing the next generation of engineers.

Your task:
Once I provide a company link, draft a concise, tailored outreach email that:

Opens by recognizing the company’s achievements or values.

Introduces our team and mission.

Shows the alignment between us and the company.

Explains the benefit of partnering.

Closes with a clear next step.

I will paste the company link next.

Jada-Buchalter, 2025-09-17:
Company: Polaroid
Contact: partnerships@polariod.com

Email (When I copied the email here, it got rid of the paragraphs and spacing):

Hi Polaroid Team,
I’m reaching out because Polaroid has always stood out for capturing moments and memories.
I’m with UMSAE Formula Electric (Polar Bear Racing) at the University of Manitoba. Every year our students design, fabricate, test, and race a fully electric formula-style car. We’re driven by sustainability, creative engineering, and hands-on learning. Our goal is to push EV tech forward, while equipping future engineers with real-world skills.
Here’s where I see strong alignment between Polaroid and us:
Tangible innovation: Polaroid delivers physical moments and bridges that with modern design. We build physical machines in the lab, merging theory and fabrication in tangible, high-performance form.
Preservation & sustainability: Just like Polaroid preserves memories in analog form, we work to preserve the planet by designing more efficient, electric vehicles. Sustainable materials, efficient energy use, and clean power are central to our work.
Creative engineering & education: Polaroid inspires creatives to experiment with format, exposure, and aesthetics. We inspire engineering students to innovate, test, iterate, and solve challenges under real-world constraints. We also hope to capture moments with our racecar with your cameras.
Partnering would bring benefits like:
Polaroid gets visibility among engineering students, tech enthusiasts, and sustainability-minded communities locally in Manitoba and at international competitions.
Association with clean transportation innovation and hands-on STEM education adds depth to your brand’s story of creativity and authenticity.
Unique content opportunities: behind-the-scenes of engineering and design, photo documentation of prototype builds, capturing design progress in Polaroid format (analog meets EV tech) stories your audience will love.
We would love to explore possible sponsorship packages or joint creative content ideas which we can discuss. I can share a few proposal sketches ahead of time so we maximize our communication.
Looking forward to hearing your thoughts,
Jada Buchalter
UMSAE Formula Electric Member
UMSAE Formula Electric / Polar Bear Racing
buchaltj@myumanitoba.ca

Jada-Buchalter, 2025-09-17:
Company: Red River Coop
Contact: memberservices@rrcoop.com

Email (When I copied the email here, it got rid of the paragraphs and spacing): 

Hello Red River Co-op,
I’m reaching out on behalf of UMSAE Formula Electric (Polar Bear Racing) which is a student-led team at the University of Manitoba building a fully electric formula-style racecar each year. Our mission is rooted in sustainability, engineering innovation, and hands-on education, and we design, manufacture, test, and compete in international events.
Here’s how I believe our values align strongly with those of Red River Co-op:
Local Investment: Our work is based in Winnipeg, and all of our design, fabrication, and testing happen locally. We believe in developing talent here, and putting engineering and sustainable technology on the Manitoba map.
Environmental Sustainability: Building electric vehicles aligns directly with reducing emissions, embracing clean technology, and driving the transition toward more sustainable transportation.
Community & Education: As a student organization, we provide hands-on learning, mentorship, and real engineering experience. We help prepare the next generation of engineers who will lead sustainable innovation which is something I think Red River Co-op cares about, given your environmental and community giving initiatives.
Partnering with UMSAE offers Red River Co-op several benefits: 
Exposure in local & national engineering/EV communities through competition, media, and university channels.
Association with innovation, sustainability, youth development enhancing your brand as a supporter of clean tech and education.
Opportunity to directly contribute to clean transportation innovation here in Manitoba, which complements Co-op’s environmental sustainability efforts.
Would you be open to explore a partnership between Red River Co-op and UMSAE? I can pull together some specific sponsorship packages or project ideas to show what we could achieve together for our 2026 electric racecar.
Looking forward to discussing possibilities with you,
Jada Buchalter
UMSAE Formula Electric Member
UMSAE Formula Electric / Polar Bear Racing
buchaltj@myumanitoba.ca