UMSAE-Formula-Electric/electrical #44: LT Spice Submissions

MitchHewitt, 2025-09-21:
## Task Description
Paste a screenshot of your LT spice simulation with the NE555 timer showing you have a 50% duty cycle on your output waveform 
50% duty cycle means that the signal is high for the same amount of time that it is low

## Expected Outcome
Screenshot of schematic in LT Spice
Screenshot of output waveform in LT spice (Voltage output)

## Additional Context
Please paste the screenshots in a comment below this issue with your name

nistood, 2025-10-26:
My name is Nisat, here's screenshots of my LT spice.

[image]
[image]

Keo-Tran, 2025-10-28:
[image]
Keo Tran - This is my ltspice tutorial

karamveerd, 2025-10-28:
My name is Karamveer Dhillon. This is my LT Spice tutorial. 

[image]

AlishaI06, 2025-11-02:
My name is Alisha Irfan and here is my LTspice output.

[image]xxxx