UMSAE-Formula-Electric/electrical #45: Altium Schematic Submissions

MitchHewitt, 2025-09-21:
## Task Description
Recreate the LT spice schematic but in Altium adding on necessary components for the final project

## Expected Outcome
all parts have a footprint and link to manufacturer part number
all parts have values listed 10k resistor is labeled as a 10k resistor and so on

## Additional Context
Post screenshots as a comment below this issue, write your name in your comment

Keo-Tran, 2025-11-09:
[image]

[image]

My name is Keo Tran and these are my altium schematics for the onboarding PCB.

nistood, 2025-11-09:
[image]
[image]
My name is Nisat Sneha and these are my Altium schematics.

luke88414, 2025-11-23:
[image]

[image]

Luke Enns schematics

AlishaI06, 2025-11-30:
[image]

[image]

My name is Alisha Irfan, these are my schematics.

tarandeepbrar123, 2026-01-05:
[image]
[image]